ChristophGeske/ARCoreInsideOutTrackingGearVr #1: Are the S6, S8 and all the other GearVR capable devices supported?

ChristophGeske, 2018-02-05:
I am not sure if the S6, S8 or one of the other GearVR capable devices are supported. Please share your experiences if you have one of those devices.

damiandabcom, 2018-02-06:
I have tested the "room" and "trivial" scenes with S8 and it works only as normal VR app, no tracking. The default HelloAR working correctly.

ChristophGeske, 2018-02-06:
I encounter the same issue on my phone (S7) before. The GearVR is loading the game just fine but the positional tracking doesn't work. I suspect that the ARCore is stopping in the background for some unknown reason. Simply restarting the game was not resolving the issue for me but restarting the phone helped. Could you try that to rule out that this bug is not responsible for the lack in compatibility.

ChristophGeske, 2018-02-24:
I just uploaded an ARCore v1.0.0 version that might be compatible with S8, S8+ and Note 8 additionally to the already tested S7.

closed 2018-02-26 by ChristophGeske