ChristophGeske/ARCoreInsideOutTrackingGearVr #3: Please add Samsung S6 support!T_T

01GOD, 2018-03-12:
Hi, 

Really been trying to get it to work on the S6. Phone is rooted and installed special apps using magisk and so on, yet still isn't functioning. 

Thanks in advance for making it funciton on the S6!!!:) Plan to make some AWESOME apps!

ChristophGeske, 2018-03-12:
I did the same when ARCore was still only available for the S8. I rooted my S7 and installed the batman cutom ROM which made my phone apear to be a S8. This worked for me at the time and I got inside out tracking to work and could start developing for it. But it was almost unusable because tracking was very bad I guess due to the different camera in the Galaxy phones giving the ARCore app a hard time.

Therefore I can not recommand installing a custome ROM just for getting ARCore to work. If it works it probably works badly or not at all. If you don't care how hard it is check out this converstion https://github.com/tomthecarrot/arcore-for-all/issues/68

I hope the S6 will be supported by ARCore in the future but I can't tell you when this day will come. As soon as the S6 is officially supported by ARCore you will be able to use the inside out tracking apps listed on this git hub page. Making a GearVR app to run without overheating after a short time is tricky even on the S7 and on the S6 it will be even harder. That said I think it should be possible to run inside out tracking apps on the S6 when keeping the environment simple.

Heath123, 2018-09-12:
You mentioned uSenseAR in the to do list. Looking at the site it seems to support tracking on more phones. Could that be added if you get the beta?

ChristophGeske, 2018-09-12:
I applied for the developer licence for uSenseAR but I did not recive the Unity plugin jet or maybe the plugin is not ready to be published. 
But Manomotion published a working version which I got to run on the gearVR with positional tracking working. In principle hand tracking in combination with positional tracking should work now with the help of the manomotion ARCore plugin. I am short in time and just did a few tests so I can't share any working code or tell you if it finally will work but it looks promissing. If you need it in next few months I would recommand to try it out yourself.

Heath123, 2018-09-15:
I can't program in Unity really. Please release the Manomotion thing soon because I've only seen videos and I'd like to see how well it works.

ChristophGeske, 2018-09-15:
I feel the same and I am very courious myself how well it works on the gearVR.
 
I have developed a version using markers for hand tracking which require you to hold a picture in your hand but tracking is not good enought compared to the controller. Also the extra burden of printing and attaching a marker to your hand is just to labore intensive. 
But the biggest negative is that the field of view of the camera is very small limiting the area you can do interactions in. 

I think the small field of view will be the biggest disadvantage for the hand tracking version with manomotion as well. Nontheless maybe some games where your head is mostly still and you look at your hands all the time could be interesting. 

In short I don't think hand tracking will be a good alternative to the controller of the current GearVR and I don't think the Manomotion version will come close to the hand tracking quality of established hardware manufacturers like leap motion. But it is an interesting project and I will keep you updated.

I think the upcomming Santa Cruze with it's wider field of view would be better suited for hand tracking and also brings enought performance.

Heath123, 2018-09-15:
Maybe if you could stick a fisheye 180° lens onto it. By the way I don't have a GearVR.

ChristophGeske, 2019-03-12:
The Vuforia tracking should works with the S6 but the quality at this moment is not very good I might be able to improve it a little bit with the filter I am developing for ARCore but it won't be the same as using ARCore for the forseeable future. The VuforiaTracking(Experimental).apk should run on the S6. Maybe a new Vuforia updates will improve the tracking quality but for now only poor quality positional tracking can be achieved with the S6.