ChristophGeske/ARCoreInsideOutTrackingGearVr #7: Example apk files install under same package name

laurenweinstein1, 2018-05-30:
Hi. This is great work and WhiteRoom is working fine here on an S9+ (actually using a DayDream 2017 headset). However, it appears that both WhiteRoom and WhiteIsland are building to the same package name, and so both cannot be installed simultaneously without additional modifications. Could these be separated out with different package names for simultaneous install? Thanks!

ChristophGeske, 2018-06-01:
Thank you for bringing this up. I uploaded a new .apk which should solve the issue. You can also uninstall the WhiteRoom app and then install the WhiteIsland app if you don't wanna redownload or resign the apk.
By the way how do you get the game to run on the Daydream headset. The only way to run GearVR apps on daydream I know of is by owning a GearVR installing the GearVR software and than turn on developer mode in the OculusService settings. 
Is that how you did it or do you found a new solution?

laurenweinstein1, 2018-06-01:
Hi. Thanks for the changes to the apk files! To run Gear VR apps on
Daydream, using for example an S9 phone:
1) Install the Oculus services
2) Get the ID for the phone, feed it to Oculus, and get back the signing
file
3) Sign an app, then Gear Developer mode can be enabled on the phone (which
it won't do otherwise)
And that's pretty much the ball game. With Dev mode enabled, at least the
S9 is a good match for the Daydream 2nd gen (2017) headset. And of course
you need a Gear controller to do anything useful (under $20 on eBay)
The tricky part with Daydream for AR or 6DOF is exposing the camera on the
phone. Here's how I solved that without cutting holes in the headset flap
(I'm actually now on my 2nd gen strap that doesn't use cable ties). Thanks
again!
https://lauren.vortex.com/2018/05/20/using-googles-daydream-vr-headset-for-augmented-reality-and-positional-tracking-applications

laurenweinstein1, 2018-06-01:
I have them both installed now. WhiteRoom remains solid as a rock.
WhiteIsland is not usable. The two images appear one after the other with a
delay of some seconds between, then both images begin to flicker badly and
continuously -- and any phone movement causes massive image smearing.

ChristophGeske, 2018-06-02:
Thanks again for letting me know. A  button in the camera setting was set wrong and everything should work fine now. 
I also worked on the tracking and the WhiteIsland app is now even smoother than before.

laurenweinstein1, 2018-06-02:
My pleasure. Thanks!

closed 2018-06-04 by ChristophGeske

laurenweinstein1, 2018-06-04:
WhiteIsland is working fine now. Thanks! I have had some instances with both WhiteRoom and WhiteIsland when the phone suddenly seemed to go into standby. It would immediately come out of standby and behave normally again when power is pressed again. This doesn't seem like an overheating shutdown, might be the physical button getting pressed accidentally in the headset, but I thought I'd mention it. Not sure what's actually going on.