ChristophGeske/ARCoreInsideOutTrackingGearVr #8: WhiteRoom and WhiteIsland occasionally go to S9+ "Standby"

laurenweinstein1, 2018-06-04:
Both apps are now functioning excellently. However, I am noticing that sometimes (always? - unsure!) after running for a couple of minutes the phone goes to standby. That is, the screen just goes black, and will come right back to where it was if the power switch is pressed again. I initially thought that the power switch was accidentally being pressed by the system I use to hold the phone in the open Daydream headset, but now I'm doubting that. This does not seem like a typical heat-caused shutdown -- for which I'd expect a warning and the phone completely shutting down -- not just entering standby and coming right back when the power switch is pressed again. It's quite possible this is some heat-related aspect with which I'm unfamiliar, but I don't have enough information at this point to judge what's actually happening.

laurenweinstein1, 2018-06-05:
I just had a thought on this. I was about to start monitoring temperatures, but it occurs to me that this might actually simply be a screen timeout occurring. I need to test this directly but is it possible that the apps as built are not keeping the screen awake (overriding normal timeouts) in some situations (e.g., when phone is being used in a Daydream headset instead of an actual Gear headset?)

laurenweinstein1, 2018-06-05:
Testing suggests that indeed the screen is timing out, that whatever keepawake function is present is not working properly. Both WhiteRoom and WhiteIsland will go black (and the background music will stop) when the current display timeout hits, unless some other activity keeps them awake.

ChristophGeske, 2018-06-05:
I am not aware of this phenomena on my device. 
Unity recommends to turn of sleepTimeout on applications using accelerometer as the main source of input. 

https://docs.unity3d.com/ScriptReference/Screen-sleepTimeout.html

Have you tried changing the display settings to a longer screen on time. (Settings/Display/Screen timeout) If you are not using a controller this might be the reason. If that's indeed the case I will fix that as soon as possible.

laurenweinstein1, 2018-06-05:
When I was actually using Island I was using the controller to fly around,
and I got the timeouts. Then I tested both Room and Island sitting on a
table (no controller). Both went screen dark and music stopped at my device
timeout interval (I tested both one minute and two minute display timeouts
-- both triggered right on the nose).

ChristophGeske, 2018-06-10:
Fixed by latest update.

closed 2018-06-10 by ChristophGeske