ChristophGeske/ARCoreInsideOutTrackingGearVr #9: Detecting real world barriers

laurenweinstein1, 2018-06-24:
Is there a practical method to use signals or cues from the camera to detect and warn (audibly or visually or both) of real world hazards -- e.g. walls or other obstacles?

ChristophGeske, 2018-06-26:
That's a fantastic idea and I started working on it yesterday. By measuring the distance between detected feature point and head position it should be possible to only show the boundary when coming to close to an object.

laurenweinstein1, 2018-06-26:
Excellent! Thanks!

laurenweinstein1, 2018-06-30:
Will try this ASAP. Thanks!

Heath123, 2018-09-16:
Could you make it work on the Cardboard version and show detected walls, and also only show when you get close to it? I think that would improve it.

Heath123, 2018-09-16:
Maybe making the points bigger too would help.

ChristophGeske, 2018-11-30:
In this unity talk a good solution is presented which I might use in the future to capture the surrounding making the boundary detection better. https://www.youtube.com/watch?v=jD1gBuVRtZs