ChristophGeske/ARCoreInsideOutTrackingGearVr #10: Arcore tracking errors

jondyne, 2018-07-01:
Firstly thank you for the amazing work you are doing with this!
I'm trying to use https://8thwall.com/ to access ARcore and to enable cardboard integration
having some trouble with tracking - see https://www.youtube.com/watch?v=KFUErLyLrbQ
is this something you also experience?

ChristophGeske, 2018-07-01:
I haven't worked with 8thwall before but there are two things I would check.

First:
When the ARCore camera is rotating it is as if the camera is around 10 cm in front of your head this could lead to some distortion effects. In my app I use a second camera to follow the ARCore camera. I moved the second camera back roughly 10 cm so that the camera "lense" is aligned with the ARCore camera position point . With ARCore camera position point I mean the point that is described by the x,y and z positions of the ARCore Device. 

[image]

Second:
I only use arcore for tracking the position in space and let my smartphone/GearVR handle head rotations. I don't think this is the problem but it could be tested quickly to make sure. 
Try to set the ARCore camera to "position only"
[image]

In this picture I added a block which helps me see what the camera is seeing. The green and blue arrows mark the ARCore camera position point and the white lines show my second camera. 

Why your box moves past the lens outline into the black area I have no idea.

jondyne, 2018-07-01:
Thanks for the detailed response. Will check!

closed 2018-11-29 by ChristophGeske