ChristophGeske/ARCoreInsideOutTrackingGearVr #13: Viking Village (Daydream) is not tracking at all

laurenweinstein1, 2018-07-11:
The Daydream/Cardboard build of Viking Village is not tracking for me. There is a static image but it does not perceptibly change with head or body movement. The Daydream/Cardboard version of BoxyRoom appears to be working properly at least in a short test so far.

ChristophGeske, 2018-07-11:
The vikingVillage.apk is the GearVR version and the VikingVillageCardboard.apk is the Daydream/Cardboard version. They booth work on my S7. 

Is the VikingVillageCardboard.apk the one that doesn't work for you?

I got some feedback on the VikingVillageCardboard.apk already and it seemed that there is no problem for other users. 

When nothing is moving it could be a sign that ARCore is not working correctly. I experienced the ARCore crashing in the background before which made the screen freeze just like you describe. Could you try the Cardboard version (VikingVillageCardboard.apk) again after you have restarted your phone. Please let me know if that fixes the issue for you. 

And thanks a lot for letting me now when something is not working this is very helpful.

laurenweinstein1, 2018-07-11:
My pleasure. No go after phone restart. VVCardboard starts and just freezes on the initial image. I'm pretty sure that ARCore is running, since I can immediately afterwards run Daydream/Cardboard BoxyRoom and it works fine. Also, as I mentioned, the Gear VR version of VV is working fine. This is all on my S9+. Thanks!