ChristophGeske/ARCoreInsideOutTrackingGearVr #14: A couple of thoughts re Viking Village (Gear VR version)

laurenweinstein1, 2018-07-11:
So this is working on my S9+. It is somewhat laggy -- that is, some jitter in tracking, and blue background can take time to fill in with the image as you move your head or body, with new blue background appearing in other areas. The same tracking issue I've noticed with the other apps is here also -- sometimes the tracking motion with body movements is backwards -- that is, when you physically move toward an object, you end up moving away from it in VR instead. The detail in this app is really very impressive!

ChristophGeske, 2018-07-16:
Hi, thanks for testing the viking village GearVR version.

Let me address the issues one by one.

1. laggy. In short I am not sure what causes the lag on your device. I experienced no lag while using this experience with my S7 so I run an analytics toll (OVR Metrics tool from Oculus) with the app running in parallel to see exactly what the performance demands are. Turns out the app runs easily 60fps on my S7 so jitter can not be caused by performance problems. I would try to restart the phone and try again in this situation. 

2. the blue background. This is expected behaviour. The viking village uses the so called seurat pipeline which can capture high end environments and simplify them so they even run on a mobile device. The big trade off with this technique is that only a predefined area can be explored without seeing distortions or blue areas. Here are some informations about the working s of this tool https://developers.google.com/vr/discover/seurat. 
So graphics are great in a limited area the area between the beginning of the bridge and the wooden gate should be fine. In this video you can see how the scene was made and the blue box is the area you can move in freely without seeing blue areas or distorted meshes. https://www.youtube.com/watch?v=CpZ94YDufqk&feature=youtu.be

3. The tracking being backwards. This is a limitation I am aware of and which is present in every experience. I hope I will find a good solution with the next big update (Spoiler alert: 6DoF controller tracking) but for now it can be avoided by holding the phone very still in the beginning and tracking the room slowly. It happens because two different head tracking modes are used which work independently from one another. The standard head tracking coming with the GearVR and the positional tracking by the camera. They booth measure the rotation independently and if the camera gets covered or looses tracking but the GearVR continues to measure the correct rotation they can become misaligned and the positional tracking by the camera moves differently from the GearVR rotation.
Therefore if at the beginning the gearVR and the camera look in the same direction and the camera has a good understanding of the room this bug can be eliminated completely. This is not an optimal solution and I try to fix it.

laurenweinstein1, 2018-07-17:
All understood. Thanks again!

Heath123, 2018-09-15:
For the tracking being backwards I found this on the Internet:
"Yes, I'm using ARCore and GVR together and see the same thing. I now track the ARCore Y rotation and the VR Camera Y rotation, and when the differ by about 5 degrees (make it easily configurable) I rotate the Player go (under which the camera resides) and then the VR world aligns with the AR world and you can walk around accurately."
Could that help?

ChristophGeske, 2018-09-15:
Yes, this is close to the solution I currently use in my latest version (not uploaded yet). I will upload it together with the other improvements I am currently working on.

closed 2018-12-04 by ChristophGeske