ChristophGeske/ARCoreInsideOutTrackingGearVr #15: IDEA: Idea for better 6DoF controller tracking

Heath123, 2018-09-15:
[image]
I put this as an issue because there isn't really anywhere else to put it. It is not an issue.
Could something like this but with markers 
attached and a Gear VR controller help with the FOV problem by increasing visibility to the camera?

ChristophGeske, 2018-09-23:
Your invention might help in some cases but e.g. turning your wrist slightly would already result in large tracker movements. I don't think there is an alternative to tracking the hands directly. I think combining both data of the controller and the camera is the right way to go. This way you have real 6dof hand tracking in front of the camera and a simple 3dof tracking when you leave the field of view of the camera. I also would like to avoid extra hardware beside the standard controller. Your second idea of adding a fisheye might work but would require extra hardware and adjustments to the algorithms behind ARCore by Google. I therefore thing that fisheye is not a possible solution.

Heath123, 2018-09-23:
Maybe it's not a good idea but I just thought it could at least help for testing.

Heath123, 2018-09-23:
With the fisheye lens, maybe if someone can modify ARCore and some hand tracking software and apps get released that work with it (not on the Gear VR Oculus store though) and Samsung will put it on the next Gear VR and Google will officially modify ARCore and apps will get released using it officially.