ChristophGeske/ARCoreInsideOutTrackingGearVr #20: No activity in MoverMarkerTracking

laurenweinstein1, 2018-12-14:


laurenweinstein1, 2018-12-14:
Just tried MoverMarkingTracking. Pass-through video looks fine, and Vuforia overlay ID is present. But that's all. Nothing I do with the controller or with my hands anywhere in front of the goggles has any effect at all. Am I missing something? Thanks!

ChristophGeske, 2018-12-14:
The app allows for the tracking of a moving marker. 
The marker that I use is a playing card and a picture of the card can be found here: https://github.com/ChristophGeske/ARCoreInsideOutTrackingGearVr/blob/master/MovingMarkerTracking/King.jpg 
When you open the link on your Laptop and look at the image through the headset the app should recognise the marker and display some cubes around it. That's all the app can do not very spectacular ;). 
If your print out the marker you could in principle attach it to your hand or the controller and track the hand in 6DOF independently from the headset.    

This app can only track this one marker (King.jpg) but if you like to track a different marker/picture or even an object you can change that in the project settings. Vuforia is the software that is able to recognise the image/marker even when it is moving. 

ARCore is also able to detect images/markers but when you move the image/marker ARCore loses tracking. This means Vuforia is currently better in image tracking than ARCore and that's why I use it. I am sure ARCore will support moving marker tracking in the future but right now it seems that using Vuforia is the best option.

Hand tracking completely without markers is possible with the Manomotion software but this is a different project. The Moving Marker tracking project uses Vuforia and is only able to track images/markers.
 
The hand tracking functionality provided by the Manomotion plugin has still some big issues:
1. Performance is still very bad. I think this is due to the Manomotion image processing requiring a lot of resources. 
2. The software uses the hand color to distinguish the hand from the background. The hand tracking is quickly lost when the background contains similar colors.

In short moving marker tracking works ok but needs some more work to be useful. Naked hand tracking is far from ready and has some big issues which need to be solved first.

closed 2019-06-17 by ChristophGeske