ChristophGeske/ARCoreInsideOutTrackingGearVr #22: Can ARCore InsideOut Tracking be combined with GearVr, while streaming with ALVR, to achive 6DOF?

dpluigi, 2019-02-07:
Hi,

This s great project and running WhiteIsland_ demo was very smooth.  Thank you to the developers.  I am just getting my feet wet with AR and VR, and I am learning a lot so far.

I am using a Flight Simulator, Condor Soaring, by injecting it to Revive and Streaming from my PC (win 10 64Bit, i7-7700K, 32GB, RTX 2080) to Gear VR (with Samsung S9) with ALVR.  It is smooth and functional at 60FPS.   It is a native 3DOF but lacks the 6DOF (5DOF really needed) to be able to lean forwards (to look at panel instrument, PDS or maps) and sideways to better see over the edge of the glider's cockpit (not as important).  Note that up and down movements are marginal.  Note also, that the requirement is mostly for the front facing position (looking over shoulder +/90, up and down +/-80, leaning forward or side to side  by up to 30cm from neutral position).  Also, so far 1 or 2h session with ALVR doesn't make the phone heat up much and drain battery too quickly.

Is there a way to combined ARCore Inside Out Tracking with GearVr, while streaming with ALVR, to achieve 6DOF?

If so, how can I integrate it?

Would fixed marker/stickers help smooth and consistent tracking and would ARCore affect adversely the mobile phone CPU load and render image fps received)? 

Thank you in advance for your help and clarification.
Donat

ChristophGeske, 2019-02-08:
Hi Donat.

> Is there a way to combined ARCore Inside Out Tracking with GearVr, while streaming with ALVR, to achieve 6DOF?

The developer of ALVR released a 6DoF version which can be found on his release page (https://github.com/polygraphene/ALVR/releases) which allows for positional head tracking. 

> If so, how can I integrate it?

You probably want the Flight Simulator to run on the ALVR ARCore version and if it already runs on the normal ALVR version switching to the ALVR ARCore version should be straight forward but for details you should check the ALVR github page. If the Flight Simulator supports 6dof with Rift/Vive it might work but agein the ALVR github page is the right addres 

> Would fixed marker/stickers help smooth and consistent tracking and would ARCore affect adversely the mobile phone CPU load and render image fps received)?

A feature rich background is required for the tracking to work at all and helps to avoid loss of tracking. If more features can improve tracking and to which point is an an interesting question but I have used ARCore on different surfaces and have not experienced a noticeable improvement of tracking when using a more contrast rich background. I therefore think tracking can not be improve by adding extra markers to your room. I think a contrast rich environment like a messy room is the best you can do. 
The ARCore team should know for sure so asking on their page might give you a better answer.

Personally I am just interested in improving the tracking in a normal room without extra markers to keep the set up as simple as possible. In my next release tracking will be improved further based on some filtering I do to the ARCore tracking data this will reduce lag and remove jumps and jitter further so should be a nice update. 
I have never used the ALVR app myself (I don't have a vr ready pc to test it 😢) so I don't know how good the tracking is but I might check it out in the near future and have a look if I can improve it.
Improved tracking on this project can be directly transferred to the ALVR project since we are both open source.😃 

ARCore affects the performance but the S9 should be able to stay cool even when ARCore and VR run together. You should try out the ALVR-ARCore version and see if your phone stays cool using it over longer periods of time.

dpluigi, 2019-02-08:
Thank you for the detailed answer.  I could not find any details  about the ALVR 6DOF on the release page you linked.  I will check with the ALVR forum.
 Your point about normal room condition is a good one and from the test  did with WhiteIsland, inside out tracking worked great on a low lighting hallway I followed out of my small living room.  If I run into issued I'll followed on the ARCore page.

Thank you again Christoph for a quick and thorough answer.
Donat

ChristophGeske, 2019-02-09:
> v2.1.0
> @polygraphene polygraphene released this on 20 Jun 2018 · 112 commits to master since this release
> 
> Support sound streaming from PC to headset. f834d2f
> USB tethering support.
> ALVR client now send the broadcast to all network interfaces except for mobile network.
> Fix black screen issue after re-launch app.
> Automatically uninstall other ALVR driver on launch. 9652e0f
> Add ARCore capable build. Only works on Galaxy S7, S8 and S9.
> More aggressive auto connect. a68ef1a #35
> Client is available here.
> Please use GithubNormal apk or Oculus Store for update from old version.
> 
> Use ARCore version only if you want to use ARCore positional tracking.
> ARCore version and Normal version are different app, so that you can install simultaneously.

Quoting from the release page. I am not sure if v2.1.0 is the only version that allows for ARCore tracking or if all releases have this functionality.

dpluigi, 2019-02-11:
Hi,

I was able to test ALVR (2.3.1) + ARCore, which is not available on Oculus store but has to be manually downloaded, signed and then installed.  Note however that this latest client apk of ALVR with ARCore to be installed on mobile headset was not downloaded from the link you first tuned me on: https://github.com/polygraphene/ALVR/releases
but from :  https://github.com/polygraphene/ALVRClient/releases
ALVR 2.3.1 client + ARCore is compatible with both ALVR Server (installed on PC) 2.3.0 and 2.3.1.

It is possible to "slide" up and down during tracking (vertical offset) with the Gear VR side trackpad (moving finger up and down) and "slide" in and out of VR to forward Camera view (moving finger horizontally on the pad).  This is quite useful for now.

All in all, the tracking was smooth and the rendered view FPS stayed at the nominal values of 60fps.  My cellphone did warm up during the session, so long session might not be practical.
I only did a couple of tests, and it is promising.  In the end, apart from zooming in and out of the instrument panel and PDA while seating cockpit, which is made possible with ARcore, the regular ALVR client might be sufficient and more stable.

I experienced mild VR sickness since I was also experimenting with VR Depth parameter in Condor VR setup and I flew using keyboard arrows (without joysticks and rudder pedals),  in other words with lots of PIO  (pilot induced oscillation) (~;

I will search through ALVR forum for more details about view centering and the possibility of setting boundaries to 6DOF, or at least add some constraint, to help stay in cockpit and/or minimize positional drift.  I did several quick repetitions of left-right head movements and little by little my position drifting further back from my seating position (in both real and virtual world - would marker help in this case is the field of view is contract or feature poor for consistent tracking?  Zooming in and out was my first goal, leaning in sideways to the edge of the cockpit is an extra.

Thank you again Christoph for putting me on the right track.

ChristophGeske, 2019-02-11:
Not sure about the reason for the positional drift you are experiencing. At the beginning ARCore needs a few seconds to detect features if you move very fast at this stage the tracking can be lost and the camera can move away. 
Also if you look into a new direction ARCore has not seen before to quickly it might lose orientation and starts tracking again at this new location resulting in position jumps.

The solution to booth issues would be to look slowly once around your room for the first 10 seconds so ARCore has seen it all and if you make fast movements later on it recognises the stored features. 
After the initial scanning fast movements should not result in loss of tracking anymore.

There is also some very small drift over long periods of time and Arcore has a feature called 'anchors' which can be used to eliminate this drift. But this drift is very small I dont think your issue is caused by this very small drift.
Try to move slowly at the beginning and see if it helps solve the drift issue. Otherwise we might need to look at 'anchors'. 

Using a marker would be possible using the 'augmented images' function. Implementing it should be easy but again I don't think this is cause of the problem you are experiencing.

I will look for drift in my further developments. Keep us updated if you find a solution or the reason for the drift.

dpluigi, 2019-02-11:
Hi,

A quick clarification. I first looked straight ahead for most of the head poses, glancing side to side slowly at first. I will hold a steady, straight ahead pose next chance I get to try it.  The question is when is ARCore initialization occurred?
A brief message might help the user know when to hold a pose for ARCore to capture the neutral / original tracking reference. 
Note that the streaming starts some time after the connection is established between ALVR client and server pair.

On a side note, is it ok with to continue posting here about ALVR / ARCore?

It looks like I'd benefit to learn about the algorithm and process in both ARCore and ALVR from you already pointed out.

The drift I noticed was induced after I noticed small amount of drift, which occured slowly at first, looking over one shoulder to another. So I decided to exacerbate it by doing a quick repetition. It was not a typical head movement.  It is also worth noting that I a seating 4ft in front of a 42" 1080p TV, with a glossy black plastic frame.  I was thinking of putting 4 white market on the frame corners.

The question, now is how can I reset the position to the initial neutral pose with a keystroke/button on my keyboard,  joystick or controller?  Another could be also used to set a new neutral position (i.e. straight ahead).

Is there a way to constrain the number DOF tracking (say 4 or 5) and/or add boundary limits?

I noticed options to  define offset parameters and programs to initiate on the  Server side of ALVR (on PC).  Once I get a couple more tests done it'll be clearer in my mind and I'll post on ALVR Forum as needed.

Have a good week.

ChristophGeske, 2019-02-11:
> A brief message might help the user know when to hold a pose for ARCore to capture the neutral / original tracking reference.

This is planned for this project (maybe in a week or so I have this feature ready), but I can not speak for the ALVR project since I am not directly working on it. 

> On a side note, is it ok with to continue posting here about ALVR / ARCore?

I can just give you some general tips I encountered working with ARCore and VR. But I am happy to help or include it in future update if the problem is a general one. The idea behind this project is to bring positional tracking to as many devices as possible and developing all the necessary features so using it is safe and comfortable. This way other developers can use it in their projects (e.g. ALVR, Quake and so on). 

> The question is when is ARCore initialization occurred?

After the pop up asked you to enable ARCore or when the unity loading screen is over and you start seeing the scene.

> Note that the streaming starts some time after the connection is established between ALVR client and server pair.

Those question I can not answer since I don't use or develop for ALVR. You might ask in the ALVR forum instead and the people using it might know.

> It is also worth noting that I a seating 4ft in front of a 42" 1080p TV, with a glossy black plastic frame.

You can try different backgrounds and see if it helps. In my house white walls are a no go. Also when something is moving on the screen this could be a problem.

> The question, now is how can I reset the position to the initial neutral pose with a keystroke/button on my keyboard, joystick or controller?

The white island GearVR app has a 'reset view' function activated by touching the touchpad. This only resets the view direction to align with the ARCore camera but doesn't set you back to the starting position/rotation. Resetting to the starting position is possible and would be similar to the reset 'view function' slightly changing the code. 
I do not know if ALVR has this functionality implemented this project has no such functionality yet. 
I will add it as a feature request it might be useful for others too but it is not a priority functionality at the moment. #23 

> Is there a way to constrain the number DOF tracking (say 4 or 5) and/or add boundary limits?

Yes, this should be possibly but I do not know if the user would like it. It might make the movement feel wrong. In general you don't  want to force the movement because he might get angry, sick or booth. 😉 But if you think it is a good feature you can have a look at the scripts in this project and get started.

> I noticed options to define offset parameters and programs to initiate on the Server side of ALVR (on PC). Once I get a couple more tests done it'll be clearer in my mind and I'll post on ALVR Forum as needed.

Yes, this question is better suited for the ALVR forum.

dpluigi, 2019-02-12:
Hi,

I used ALVR+ARCore version 2.3.1 and experienced some drift after a while, which I corrected my hand using the "Other/Apply Offset to Player position" in the Server ALVR.  It is not ideal and I hope I can use a more direct method.
There is a view re-center option that can be enable in Controller ALVR Server menu, but it is flipped by 180deg, so no useable until it is fixed and I'd prefer to create a shortcut to keyboard or a joystick button.
Also, I had some issue where depending on how ALVR close (or crash) Gear VR controller highjack my mouse and I am stuck and can't navigate on my desktop until I kill Steam/ALVR... It looks like the VR controller setting in Steam or VRSteam needs to be twicked but I am not sure how.

ALVR+ARCore did make my Samsung S9 hot after 1 hour of use, and at the first warming I closed it and started ALVR Normal instead.

At this point if I could get a zoom in the direction of head pose (forward looking) then I would not really need 6DOF, 4DOF would suffice or 3DOF with zoom option to zoom in on instrument panel/PDA while in flight.

The Anchor point would probably be the way to go since in my case I use a set seated position and do not need full room scale tracking, and most of the time the head would scan half of the space (forward) at the very most.  In that respect NOLO tracking would be an option but I don't think it is worth it for $200 when Oculus Rift with room tracking and Touch controllers is $399.  

The forum is not as responsive on ALVR and I am wondering if I should consider having a stab at implementing these couple of features for me to try out.  It has been over 15yrs since I last programmed in C++.  I am wondering how much work it'd be to get all installed on my PC (for free) to compile test APK?

Also, I used Google AR Measure tool to test that ARCore was detecting surfaced.  Do you know if there is a test tool that can show me which ones are point used and tracked by ARCore so I can check how fares in my room setting?

Anyhow, I am hoping that the Mobile VR headset based on Gear VR + S9 would allow me to wait for better offering.  SO far it is looking promising as I have modest expectation and patience until the hardware improves.
My ideal headset:  Inside out tracking/room scale tracking capable, higher res, minimum of 120deg FOV, wireless, eye-tricking to allow foveated rendering... for less than $500-600.  One can dream or wait  couple more years?!  (~;

Thank you again Christoph for your help and explanations.
Donat

ChristophGeske, 2019-02-13:
Hallo Donat.

> There is a view re-center option that can be enable in Controller ALVR Server menu, but it is flipped by 180deg, 

Looks like a ALVR bug you might open a new issue on their github page.

> At this point if I could get a zoom in the direction of head pose (forward looking) then I would not really need 6DOF, 4DOF would suffice or 3DOF with zoom option to zoom in on instrument panel/PDA while in flight.

You could use the touchpad or the GearVR controller to archive such a functionality. Simply move the camera in the viewing direction of the camera. I use a similar function in the [MoveController script](https://github.com/ChristophGeske/ARCoreInsideOutTrackingGearVr/blob/master/WhiteIsland/Assets/NewScripts/MoveController.cs)  in the white island app you can fly in viewing direction by using the controller.

https://github.com/ChristophGeske/ARCoreInsideOutTrackingGearVr/blob/master/WhiteIsland/Assets/NewScripts/BounderyController.cs

> I am wondering how much work it'd be to get all installed on my PC (for free) to compile test APK?

Since most of your features are related to bringing positional tracking to desktop powered games these features should be implemented in the ALVR software. My project is much smaller compared to ALVR and all functionalities run on the phone keeping complexity down. Since some of your features are specific to your use case of the flight simulator you might have to implement them yourself. But the zooming feature and reducing the field of view should be straightforward and should not take to much time if you have used Unity before. 
I think the ALVR project will see a big push in the coming months when the oculus quest appears and people realise they might be able to use it with their pc as well.

> Also, I used Google AR Measure tool to test that ARCore was detecting surfaced. Do you know if there is a test tool that can show me which ones are point used and tracked by ARCore so I can check how fares in my room setting?

You can use the [Hallo ARCore app](https://github.com/ChristophGeske/ARCoreInsideOutTrackingGearVr/blob/3a8a5975aaf430e406ce0cc232ed925c76bd8348/HelloARCore.apk). The blue points are the features detected and tracked by ARCore. This app is a kind of hello world version of arcore made by google.

dpluigi, 2019-02-13:
Hi Christoph,

I already have Visual Studio Code Installed on my PC, from when I was
tinkering with Andruino.  This will only allow me to edit with
Intellisense, and debug but not compile.

What IDE/compiler package would you recommend I install to compile CS and
CPP projects and Code?
Mingw-w64  http://www.mingw-w64.org/doku.php
or Visual Studio 2017 for Windows 10\
For CSharp it looks like I might be able to compile from within with
Shift + Ctrl + B according to
https://stackoverflow.com/questions/30150022/how-to-compile-c-sharp-in-microsofts-new-visual-studio-code
Do you recommend something else?

Sorry for the rookie questions.  When I was compiling in C++ it was on Unix
from command line with g++, and very briefly used Windows based IDE like
Visual C++.

Thank you for your help getting me started.

Donat

ChristophGeske, 2019-02-13:
I think the most important tool you need to get familiar with is Unity. Coding is the easy part getting familiar with Unity takes longer in the beginning. There are alternatives to Unity like Unreal and Android Studio but I think when it comes to VR development for Android Unity is the best resource.  

Unity comes with everything you need and uses Visual Studio 2017 out of the box.
C++ is used for the scripts.
I use Windows 10 and Unity 2018.2 at the moment. 
The best resource for me when I started with Unity was [this tutorial](https://www.youtube.com/watch?v=_lP6epjupJs&list=PLX2vGYjWbI0SLoX3q1tUf3RJU_WzcLcHL) but it is a bit older and there were changes made to Unity which might result in some issues when following along. 

You might find a more recent tutorial on the Unity pages they have lots of resources to get you started and youtube is also full with people doing tutorials.

So for you to get involved:
1. Set up Unity for Android development.
2. Get familiar with Unity following along with a good tutorial
3. Download the ALVR project and change the scripts that are relevant for you.
4. Add your work on github so we can all learn from it.

dpluigi, 2019-02-18:
Hi,

I installed Visual Studio and the latest Unity 2018.3.5.f1 (Free).  Visual Studio Installer gave me some grief and I only managed today to get the last tools and component required for ALVR.  The Installer wouldf stall and fail to start and I had to delete the Installer folder and even to remove Visual Studio 2017 Community and start the install over to finalize the Install.  

This week I'll start get familiarized with the code.  First I'll try to compile the project (client and server).  Then I'll talk the feature I'll like to add (zoom and camera passthrough for the Normal ALVR).

At the moment the ALVR Client + ARCore is warming up my Galaxy S9 too much to be useable.  I just read your post about ARCore 30FPS tracking.  Do you think that the optimization of ARCore and/or reduce the tracking FPS or reduce the number of DOF would help reduce the cellphone load?

Thank you,
Donat

ChristophGeske, 2019-02-18:
I think there are 3 main areas where you can look for performance savings 
1. ALVR Client: Do you need all the functionalities or can you disable some functions. Maybe you can optimize the code to reduce the requirements.
2. ARCore: There are some features you might not need like the visualisation of the point cloud. You can disable those but you have no access to the underlying algorithms and I think they are very performant already since google took care of that. 
3. GearVR: Oculus has articles about the Unity settings which are recommended for VR. I think you can not do much more than just using the recommended settings. They also have articles on what to do when CPU or GPU are maxed out but this is often related to building a whole game on the phone. Since you stream all the images from the pc I believe you have a CPU bottleneck and optimizing the code is probably what you need to do.

I don't think you can do much with the fps since you want a minimum of 60 fps for tracking and updating the display. 
Unity comes with a profiler tool build in which allows you to see which part of the code eats up most of the performance and if CPU or GPU is responsible.
For profiling my apps I do the following: 
- enable developer build in build settings
- activate GearVR developer mode so you don't have to plug your phone into the headset so that it can stay connected with the pc 
- run the app on your phone 
- open the profiler window and choose the phone to be displayed

Understanding the profiler might require you to watch/read some tutorials on profiling in unity. 
In the end I can not tell you if this will finally work or if combining all three ALVR Client + ARCore + GearVR is just to much even for the S9. We are playing at the limits of what the phones can do.

If nothing helps you could also cool down your phone. I used wet paper or a wet rag before just sticking it to the back of the phone helped remove the heat and allowed me to run some store apps for much longer on the S7.

closed 2019-06-17 by ChristophGeske