ChristophGeske/ARCoreInsideOutTrackingGearVr #26: Detected plane moves with the camera

Zumbalamambo, 2019-04-19:
I tried the boxychardroom example. I dragged the DetectedPlaneVisualizer of ARCORE  to Player Gameobject. I have two issues with this. 

1. The detected planes move as I move the camera.

2. I tried to place andy prefab on a specific location. If I move the camera, the 3d object moves as well instead of being static in its original position.

ChristophGeske, 2019-04-19:
I don't know the answer to your question out of my head and would need to implement it myself to know for sure but as a first quick answer I would suggest you have a look at the HalloARCore example scene there it should work as expected just without VR (only AR). You can use this scene as a reference and first check if the planeGenerator script and planeVisualizer script in your scene are attached to the right game objects probably to the ARCore camera object not the player object. 
If everything in your scene is correct and the issue is still there you could start with the working HalloARCore scene and step by step add the VR features you need to find out when the issue first occurs. That's how I would proceed to find the cause of the issue. If you know what causes the problem you might be able to fix it. 

I can look into it sometime in the near future and give you a better answer. I have to look into it anyway since that is interesting for improving the guardian/boundary system by displaying the floor and walls.

Zumbalamambo, 2019-04-23:
@ChristophGeske I checked and I found that its because the field of view of both AR and VR are mismatching and hence the issue prevails. Is there any fix that you can suggest?

ChristophGeske, 2019-04-24:
You can check out this project ( https://github.com/jondyne/ARcore_mobileVRHeadset/ )
they had to deal with misalignment issues as well when they combined ARCore and Cardboard. But I haven't looked into this issue myself so not sure if that fixes all your problems.

Zumbalamambo, 2019-04-24:
i checked it previously. It has a black screen issue with the latest version of arcore . isn't there any workaround to fix this issue?

ChristophGeske, 2019-04-24:
Do you get the black screen issue when testing the apk from the linked github page?
I just checked the release apk ( https://github.com/jondyne/ARcore_mobileVRHeadset/releases ) and it works just fine on my S7 with ARCore version 1.8.1

Zumbalamambo, 2019-04-25:
@ChristophGeske I get the black screen while trying to run the code in my real device through unity. I upgraded the ARCore and ARFoundation version in manifest.json file and I tried to run it. It was just showing a black screen. However the prebuild apk files works fine. Could you please kindly suggest me a solution to fix it?

ChristophGeske, 2019-04-26:
Going with the older ARCore or Unity version should work. I am not sure what breaks the project. I am also stuck with an older ARCore version at the moment since the latest ones break some of my functionalities. Getting breaking changes when updating Unity or ARCore is very common so if you wanna be sure to get a working project stick to the versions of Unity and ARCore used by the project publisher. 
If you don't need the latest features of ARCore or Unity that would be the easiest solution. Otherwise you have to find out what causes the error and fix it in the other project.

Zumbalamambo, 2019-04-29:
@ChristophGeske thanks. Backporting worked!

closed 2019-04-29 by Zumbalamambo