ChristophGeske/ARCoreInsideOutTrackingGearVr #27: ARCore , Google VR with menu

Zumbalamambo, 2019-05-07:
Which release of BoxyRoomChardboard has got ARCore Device with menu and Google VR?

ChristophGeske, 2019-05-07:
I am not sure what you mean with menu but the [BoxyRoomChardboard project](https://github.com/ChristophGeske/ARCoreInsideOutTrackingGearVr/tree/master/BoxyRoomChardboard) uses Google VR and should contain the necessary packages to build a Cardboard app running on all phones which are ARCore ready. 
The ARCore Device is used in every project as a part of the ARCore package and you should find it in the project hierarchy when you open the project and BoxyRoom scene in Unity. 
Are you looking for a menu object that is part of the Google VR package? If something in the Google VR package is missing you can re-import the Google VR package and see if the missing menu object has returned. I might have not imported all the possible objects that come with the Google VR package to reduce the project size. 
I hope this helps if not please don't hesitate to ask agein.

Zumbalamambo, 2019-05-13:
thank you @ChristophGeske

closed 2019-05-13 by Zumbalamambo