ChristophGeske/ARCoreInsideOutTrackingGearVr #32: ARcore v1.11

VRsampan0580, 2019-09-02:
我的手机用的是ARcore最新版本，我在unity导出apk时是不是要用1.11的SDK，还是要怎样操作？因为我发现按照原来的方式导出来摄像头没有定位功能

ChristophGeske, 2019-09-02:
I used a translator for your question:
"My phone is using the latest version of ARcore. Do I want to use 1.11 SDK, or what to do when I export apk from unity? Because I found that the camera had no positioning function the way it was exported."

The uploaded apk's where build with ARCore v1.6. 

Google Play Services for AR v1.11 supports apk's build with an older SDK. 

But you are right there is an issue with the new ARCore v1.11 version not running the apk properly. I also noticed the problem that tracking is not working when starting the apk the first time after installing or after restarting the phone when having the new "Google Play Services for AR" installed. 
The same issue can be observed when I build the apk with the latest SDK v1.11. This means it must have to do with the new "Google Play Services for AR"

Tracking works as expected if you close and restart the app.  

I do not know what causes this issue, but since it is depending on the ARCore version installed on the phone and the problem is solved by restarting the app it looks like a bug to me. It could also be related to the new settings introduced with the latest SDK v.1.11. 

Thank you for finding and reporting this bug, I thought I was the only one having this issue.

VRsampan0580, 2019-09-03:
但是我重新启动应用程序后定位功能依然无法正常工作，不知道是什么问题，可能我需要用旧版本的ARcore

ChristophGeske, 2019-09-03:
I also tried to downgrade the ARCore apk to [version 1.6](https://www.apkmirror.com/apk/google-inc/arcore/arcore-1-6-18111209-release/) and I did not experience the loss of tracking when first starting the app. You could try that on your device and report what you found maybe we talk about a different issue there could be multiple reasons for the tracking not working as expected.

I am not sure what is causing the bug on my device and if all of you experience the same issue. More testing is needed maybe it can be fixed by changing the settings. If I find the time I will look into it in more detail for now I can live with restarting the app if tracking is not working as expected.